kjans123/BME590-LightDHF #7: LED bulb Question

kjans123, 2018-10-11:
Hi Dr. Palmeri, would either of these options work for the LED bulb in the flashlight?

1) https://www.amazon.com/Miniature-Light-Bulbs-Lamps-Lionel/dp/B01N97FT10/ref=sr_1_3?ie=UTF8&qid=1539102701&sr=8-3&keywords=e10+led

2) https://www.amazon.com/Miniature-Light-Bulbs-Lamps-Multicolor/dp/B01CITEJWA/ref=pd_sim_60_5?_encoding=UTF8&pd_rd_i=B01CITEJWA&pd_rd_r=306aa1a7-be91-11e8-9fc4-81eb294f983c&pd_rd_w=sNoyz&pd_rd_wg=vqYkJ&pf_rd_i=desktop-dp-sims&pf_rd_m=ATVPDKIKX0DER&pf_rd_p=18bb0b78-4200-49b9-ac91-f141d61a1780&pf_rd_r=EE83F1ARXNADPWMYMN1Y&pf_rd_s=desktop-dp-sims&pf_rd_t=40701&psc=1&refRID=EE83F1ARXNADPWMYMN1Y

mlp6, 2018-10-11:
They are LED, so yep!  Keep in mind the operating voltage range for the first one...

kjans123, 2018-10-11:
Great, thank you!

closed 2018-10-11 by kjans123