kjans123/BME590-LightDHF #23: V2 Arduino Code Delay Problems

kjans123, 2018-11-01:
Sorry for the multiple issue assigns, but I've hit (more than) a bit of a sticking point in my Arduino code. The state switches work well for brightness switching, but when the LED starts flashing, hitting the button often does nothing. I know this is because the loop is getting delayed and the Arduino is not "seeing" the button press...but I am not sure how to solve the issue. I am using a code interrupt, but I think I may have it catching in the wrong place. 

Code: https://github.com/kjans123/BME590-LightDHF/blob/master/Software/SwitchOuputsCode_V1/SwitchOuputsCode_V1.ino

mlp6, 2018-11-01:
Yep, the `delay()` command is causing the `loop()` to stall, so you're really throwing off the timing.  Check out: https://playground.arduino.cc/Code/Timer

kjans123, 2018-11-02:
I GOT IT!!!! I ended up making use of millis().

closed 2018-11-02 by kjans123