kjans123/BME590-LightDHF #30: Review of ECAD designs

kjans123, 2018-11-08:
The folder in my repo for the ECAD files is located here: 

https://github.com/kjans123/BME590-LightDHF/tree/master/Electronics/eagle_sketch%26boardV3

Thank you!

sphaerobolus, 2018-11-12:
I'm not sure where you're going with JP1, JP2 and JP3. what connects there? 

You could optimize routing. Traces here are convoluted. 

On the PCB, there are several dangling pins: Battery, LEDHOLES, IC1, Zener. 

Not a good idea to implement a device through the use of two separate footprints (switchpad1 and 2).

What's the purpose of the Zener? 

What's the purpose of R3?

LEDHOLES and BATTERY shouldn't be power resistors. If they're off-board, make them headers.

kjans123, 2018-11-12:
Make footprints for Arduino pro mini and smd switch. Put 2pin headers in at 0.1" pitch for LED and battery. Remove Zener. Remove R3. Make traces straighter.

Need to figure out how to make footprints. Make board MUCH smaller (roughly the size of the Pro Mini).

kjans123, 2018-11-14:
Hello, I updated the board and schematic with the above suggestions:

board & schematic: https://github.com/kjans123/BME590-LightDHF/tree/master/Electronics/eagle_sketch%26boardV3

the custom-made library part for the smd switch is here: https://github.com/kjans123/BME590-LightDHF/blob/master/Electronics/DIPswitchLib.lbr
I was able to find a custom footprint for the Arduino Pro Mini on GitHub.

sphaerobolus, 2018-11-14:
Looking good!

closed 2021-06-04 by kjans123